Geeetech3D/Diagram #1: Marlin 2.x Support for the GTM32 mini and pro boards

redangel1984, 2018-07-10:
It would be great if these two open source boards could support Marlin 2.x 32bit version. 
I have no idea of coding. But this would be awesome if Geeetech or users who can code would make Marlin 2.x to be compatible with the GTM32 boards. So much possibilities and features would open up for the users and also for Marlin. 
I’ve seen this issue posted on the Smartto firmware repository which in my opinion would belong here, it is not a firmware issue/feature but more like a hardware issue.

monrra, 2018-08-01:
I totally agree with you. If we can install Marlin in our boards I’m sure that the community and contribution will grow up a lot. Please take this on count, I think it’s the best improvement.

alexwoo1900, 2018-08-01:
@redangel1984 @monrra  We are rewritting smartto-eclipse(reconstruct, complete command, fix bug etc.) It may take one or two months and then we will start discussion on marlin 2.x. We still not sure how many people would like to use marlin in these boards. Is it worth to do that? The issue need more research and supporter.

monrra, 2018-08-01:
Hi @alexwoo1900. I dont know if @redangel1984 thinks like me, but I really think that almost everyone prefer Marlin for being an open source reliable firmware since a lot of time ago. In fact, you just need to look for other printers with Marlin, people are very happy with it.

redangel1984, 2018-08-01:
@alexwoo1900
I am sure If you make Marin available for this printer and board more people will use it on Marlin than on Smartto.
And I am sure too that it you won’t make Smartto stable and improve it, you will loose current and future customers. Because as for now to use Smartto is a pain in the @$$.

monrra, 2018-08-01:
Totally agree with @redangel1984.

macasero, 2018-08-01:
Hi @alexwoo1900. I think @redangel1984 is right. It would be gorgeous

redangel1984, 2018-08-01:
Vote:

https://www.facebook.com/groups/315127105604393/permalink/509082706208831/

BennyBoyW, 2018-08-01:
At this point, I believe we'd be better served with the Marlin Firmware. My A30 is currently bricked till I get chance to downgrade the firmware etc

JordiKoki, 2018-08-01:
We need this project. Marlin Firwmare is more extended in the 3d print comunity. More functions, open code, etc... @alexwoo1900

edusaul, 2018-08-01:
It would be great to have Marlin Firmware for A30 @alexwoo1900

alexwoo1900, 2018-08-02:
@redangel1984 @monrra @macasero @BennyBoyW @JordiKoki 
Hi, guys.
We have seen your posts and votes and made a long discussion about how to deal with Marlin and Smartto a few hours ago.
That's our plan:
First and foremost, we will repush newest firmware which fixed all fatal errors in a short time.
Secondly, we will continue our development on Smartto. At the same time, we will contact with marlin team and talk about how to make Marlin support GTM32 boards.
I will share our progress about Marlin supporting in this post or facebook group.
If you already have done something about it, please let us know.

BennyBoyW, 2018-08-02:
@alexwoo1900 For the re-push of the firmware that fixed the fatal errors .....
Can you please provide a timeframe for this to be pushed ?  
I am stuck at the moment on V1.39.x firmware as I am trying to get time to film some of the faults I am seeing (unfortunately, my job is not allowing me that time at the moment but I hope to get to it at some point tomorrow). Once I film these, I will be downgrading to 1.35.x firmware - unless the new re-push is going to be soon

soydif, 2018-08-02:
good morning, it is very interesting this proposal would be ideal to have that compatibility for our printers. I am also interested and I have two.

schuko1982, 2018-08-02:
It would be great to have compatibility with Marlin, apart from the one we already have with smartoo, thank you very much for listening to us and wanting to improve the product.

alexwoo1900, 2018-08-14:
Marlin team has no reply to us. But we had started to do transplant work.

monrra, 2018-08-17:
@alexwoo1900 that’s great! With Marlin, this printer will be one of the best in the market.

monrra, 2018-10-08:
Hi @alexwoo1900 . Are there any news?

redangel1984, 2018-10-08:
2 month and still nothing?

alexwoo1900, 2018-10-24:
@monrra @redangel1984  Sorry for my late reply. Our firmware team met some very tricky problems. They make all works except for bootloader and SD card part. (The main dev is working on the latter). The market team had sent some copies to some customers to test. If you guys want to join us, please contact Emma.

macasero, 2018-10-24:
I would love to test it

Geeetech3D, 2018-10-25:
https://pan.baidu.com/s/1r_M4g5Q644flxyVpDe85Lg
Contains the Arduino compiler and A30 marlin source code, as well as instructions for use. Currently, serial port printing is supported, and SD card printing is not supported.

macasero, 2018-10-25:
Can´t register without a china phone number. Could anybody upload somewhere else?

macasero, 2018-10-25:
exec: "/bin/arm-none-eabi-g++": file does not exist
Error compilando para la tarjeta Generic STM32F103V series.

any thoughts

alexwoo1900, 2018-10-26:
@macasero 
In Arduino, tool->board->boards manager
Install the target board

macasero, 2018-10-27:
Solved, compiled, tested. 
Fan are wrong. Hotend cooling fan does not work. If try to enable with gcode you turn on and off the fan at control
Box. Layer fan is allways on. Seams pin mapping needs to get review.

macasero, 2018-10-31:
did you follow this schematics to do the mapping?
https://github.com/Geeetech3D/Diagram/blob/master/3DPrinter_A30_E180/Hardware_GTM32_MINI_S.pdf

rokrodic, 2018-11-17:
Where can we get GTM32 Pro motherboard schematics?

redangel1984, 2018-11-18:
From the link above.

rokrodic, 2018-11-18:
Sure, it seems so, but I have 6 stepper motor slots. The above scheme shows only 4. I could examine Smartto sources for pins, but as it is supposed to be open source... The name above states, it's "mini". I have large one GTM32 PRO.

Btw, how's with Marlin 2? My super plate is ruined due to Z-offset Smartto bugs...
And where is the color-LCD-STM32F103 source code (I have latest A30)?

kaktusus, 2018-11-18:
GTM32Pro **vB**?
The diagram has also been published for this version.

MoeweX, 2018-11-25:
How is the review process going @alexwoo1900

Sneakers82, 2018-11-27:
> Can´t register without a china phone number. Could anybody upload somewhere else?

How did you download the firmware? @macasero  
Could someone possibly upload it somewhere else?

johne3819, 2019-01-25:
@alexwoo1900 has there been any progress?  I am considering the pro board but I'm not sure.  I had the 12V and I do not like it.  It has a lot of heat / sensor error that seem to be delays rather than misreads from thermistor.  Only occasionally does it quickly give a -19, but immediately works again.  Seems to be more of a data throughput type thing

pinchies, 2019-03-11:
Hello, I'm compiling a [list of Chinese 32-bit 3D Printer boards that support Marlin 2.0](https://samuelpinches.com.au/3d-printer/chinese-32-bit-3d-printer-motherboards/). May I ask what the status is for the Geeetech GTM32 boards with Marlin 2.0?
I recently fixed a few bugs regarding fan control (recommend Software PWM in configuration.h) so that may or may not help with the fan problems from the last attempt.

bs2000, 2019-05-07:
How about support for the M201

Vertabreak, 2019-05-17:
if this helps anyone
[A30.zip](https://github.com/Geeetech3D/Diagram/files/3191283/A30.zip)

S91C3, 2019-05-31:
I am very much interested in a Marlin 2.0 fork for the GTM32-mini. Is anyone still working on this?

InnerBushman, 2019-08-13:
To make the GTM32PRO_VB board work you need to remove -DSERIAL_USB flag in "## Generic STM32F103V ##" section of ~/.platformio/packages/framework-arduinoststm32-maple/STM32F1/boards.txt
I'm attaching my working copies of Configuration.h Configuration_adv.h and pins_GTM32_PRO_VB.h
Here's my platformio.ini entry for this env:
`
#
# Geeetech GTM_32_pro_vb (STM32F103VET6)
#
[env:gtm_32_pro_vb]
platform      = ststm32
framework     = arduino
board         = genericSTM32F103VE
monitor_speed = 250000
extra_scripts = buildroot/share/PlatformIO/scripts/gtm_32_pro_vb.py
build_flags   = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
  ${common.build_flags} -std=gnu++14
  -DSTM32F1xx
build_unflags = -std=gnu++11 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
lib_deps      = ${common.lib_deps}
lib_ignore    = TMC26XStepper
  libf3c
  lib066
  Adafruit NeoPixel_ID28
  Adafruit NeoPixel
  libf3e`
Can someone talk to platformio guys and ask them to create separate entry for this board? as this board never uses native USB for programming.
I have not make it to program via serial interface so you will have to upload the firmware the normal way or use the ST GUI demonstrator tool that Geeetech tells you to use if you mess up the firmware.
I hope this will help to kickstart the development for this board so I won't be needing to compile it myself later ;)
If i forgot about something I'll add it later. Any questions? I'll try to answer them but don't expect brilliant answers >_<'
Good luck everyone! 

[GTM32PRO_VB.zip](https://github.com/Geeetech3D/Diagram/files/3496221/GTM32PRO_VB.zip)

Avatarsia, 2019-08-13:
> To make the GTM32PRO_VB board work you need to remove -DSERIAL_USB flag in "## Generic STM32F103V ##" section of ~/.platformio/packages/framework-arduinoststm32-maple/STM32F1/boards.txt
> I'm attaching my working copies of Configuration.h Configuration_adv.h and pins_GTM32_PRO_VB.h
> Here's my platformio.ini entry for this env:
> 
> ```
> #
> # Geeetech GTM_32_pro_vb (STM32F103VET6)
> #
> [env:gtm_32_pro_vb]
> platform      = ststm32
> framework     = arduino
> board         = genericSTM32F103VE
> monitor_speed = 250000
> extra_scripts = buildroot/share/PlatformIO/scripts/gtm_32_pro_vb.py
> build_flags   = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
>   ${common.build_flags} -std=gnu++14
>   -DSTM32F1xx
> build_unflags = -std=gnu++11 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1
> src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
> lib_deps      = ${common.lib_deps}
> lib_ignore    = TMC26XStepper
>   libf3c
>   lib066
>   Adafruit NeoPixel_ID28
>   Adafruit NeoPixel
>   libf3e
> ```
> 
> In the other ZIP file are the scripts for this env.
> 
> Can someone talk to platformio guys and ask them to create separate entry for this board? as this board never uses native USB for programming.
> I have not make it to program via serial interface so you will have to upload the firmware the normal way or use the ST GUI demonstrator tool that Geeetech tells you to use if you mess up the firmware.
> I hope this will help to kickstart the development for this board so I won't be needing to compile it myself later ;)
> If i forgot about something I'll add it later. Any questions? I'll try to answer them but don't expect brilliant answers >_<'
> Good luck everyone!
> 
> [GTM32PRO_VB.zip](https://github.com/Geeetech3D/Diagram/files/3496221/GTM32PRO_VB.zip)
> [buildroot.zip](https://github.com/Geeetech3D/Diagram/files/3496310/buildroot.zip)

will this also work on the gtm32 mini s ?

InnerBushman, 2019-08-14:
> will this also work on the gtm32 mini s when i change the pinmapping ?

I haven't seen the schematic of the mini s board but if it's the same CPU and the pinmap is also touching the USB pins, this should also work. 
Why don't you try and report here? ;)

Avatarsia, 2019-08-14:
> > will this also work on the gtm32 mini s when i change the pinmapping ?
> 
> I haven't seen the schematic of the mini s board but if it's the same CPU and the pinmap is also touching the USB pins, this should also work.
> Why don't you try and report here? ;)

i tried but so far the same result as verta, hotend fan and partcooling fan are not running. 
it's a pitty that i can't find any board pin maps in smartto codes

thpparent, 2019-08-24:
> To make the GTM32PRO_VB board work you need to remove -DSERIAL_USB flag in "## Generic STM32F103V ##" section of ~/.platformio/packages/framework-arduinoststm32-maple/STM32F1/boards.txt
> I'm [...]

Hi Avatarsia,

Great job.

Was able to compile and load on a M201 without issue

Everything is working well except... the SD reader on the LCD
I'm using the standard 4 lines LCD2004. REPRAP_DISCOUNT_SMART_CONTROLLER

What LCD are you using and is it working on yours ?

Thank you so much

Avatarsia, 2019-08-24:
> Hi Avatarsia,
> 
> Great job.
> 
> Was able to compile and load on a M201 without issue
> 
> Everything is working well except... the SD reader on the LCD
> I'm using the standard 4 lines LCD2004. REPRAP_DISCOUNT_SMART_CONTROLLER
> 
> What LCD are you using and is it working on yours ?
> 
> Thank you so much

Wasn't my job. 
Wich board do you use  ?
In my understanding the libary for the mini s board is incomplete so it cant compiled well

thpparent, 2019-08-24:
Hi Avatarsia,

Same than yours : GTM32 Pro VB

Avatarsia, 2019-08-24:
I have the Mini S :)
you quote the wrong one ^^

thpparent, 2019-08-24:
> 
> 
> I have the Mini S :)
> you quote the wrong one ^^

Uuups correct. Sorry

thpparent, 2019-08-24:
> To make the GTM32PRO_VB board work you need to remove -DSERIAL_USB flag in "## Generic STM32F103V ##" section of ~/.platformio/packages/framework-arduinoststm32-maple/STM32F1/boards.txt
> I'm [...]

Hi InnerBushman

Great job.

Was able to compile and load on a M201 without issue

Everything is working well except... the SD reader on the LCD
I'm using the standard 4 lines LCD2004. REPRAP_DISCOUNT_SMART_CONTROLLER

What LCD are you using and is it working on yours ?

Thank you so much

InnerBushman, 2019-08-26:
Hi thpparent,
As you can see in my config file i also use the code for REPRAP_DISCOUNT_SMART_CONTROLLER.
I have the stock LCD/encoder/SD reader from Geeetech that seems to be based on the Discount Smart Controller, only with different ribbon cable.
The SD card seems to work ok with this config. I can start a print from the SD (haven't actually printed anything tho) and get no errors, the printer homes and starts to heat up.
What exactly isn't working? And how are you testing?

Avatarsia, 2019-08-26:
InnerBushman do you have Facebook or Discord or any way to communicate ?

InnerBushman, 2019-08-26:
> InnerBushman do you have Facebook or Discord or any way to communicate ?
InnerBushman#4784

thpparent, 2019-08-27:
> Hi thpparent,
> As you can see in my config file i also use the code for REPRAP_DISCOUNT_SMART_CONTROLLER.
> I have the stock LCD/encoder/SD reader from Geeetech that seems to be based on the Discount [...]

Hi,

Thanks for your answer.

I have the feeling that something is wrong in the SD pins definition referenced above.
Card inserted / removed is detected, but result is always the same "no media".
It fail in the three case : power ON with card, Card inserted and Remotly reading card from repetier.

If you take a look to the Hardware_GTM32_PRO_VB. pdf, the connector seems to connect SPI2 (which could be the SD Interface), in the published as part of Marlin2.0, matched PIN with SD interface doesn't seem to be logic and are not following the same logic that the onboard SD interface PAx.

Coud you confirm that your working configuration is based on the following PIN definition ?

Thank you so much

* The on-board SD card and the external card (on either SD_CARD or J2) are two
 * separate devices and can work simultaneously. Unfortunately, Marlin only supports
 * a single SPI Flash device (as of 2019-07-05) so only one is enabled here.
 */
#if ENABLED(GTM32_PRO_VB_USE_EXT_SDCARD)
  //
  // SD Card on RepRapDiscount Smart Controller (J2) or on SD_CARD connector
  //
  #define SS_PIN           PC11
  #define SCK_PIN          PC12
  #define MOSI_PIN         PD2
  #define MISO_PIN         PC8
  #define SD_DETECT_PIN    PC7
#else
  //
  // Use the on-board card socket labeled TF_CARD_SOCKET
  //
  #define SS_PIN           PA4
  #define SCK_PIN          PA5
  #define MOSI_PIN         PA7
  #define MISO_PIN         PA6
  #define SD_DETECT_PIN    -1  // Card detect is not connected
#endif

My Facebook Account is Radagast Stufly

Thierry

Avatarsia, 2019-08-27:
to my last doument it should be like this 

SDIO+DMA2 control/SPI2 
D0/M1SO | 
D1/D1 | PC9
D2/D2 | PC10
D3/CS | PC11
CLK/CLK | PC12
CMD/MOSI | PD2
CD/CD | PC7

thpparent, 2019-08-27:
> 
> 
> to my last doument it should be like this
> 
> SDIO+DMA2 control/SPI2
> D0/M1SO |
> D1/D1 | PC9
> D2/D2 | PC10
> D3/CS | PC11
> CLK/CLK | PC12
> CMD/MOSI | PD2
> CD/CD | PC7

Thanks, is it not specific to the GTM32 mini s ?

Avatarsia, 2019-08-27:
regarding my infos it will also be the same for the new a30M in the future, so no all the same

Avatarsia, 2019-08-27:
oh there get somethink lost
D0/M1SO | P8

thpparent, 2019-08-27:
To be honest, I don't recognize the format/nomenclature. Is it Marlin or RepRap ?

Thanks

Avatarsia, 2019-08-27:
copy and paste from an excel sheet ^^ 
name| pin

InnerBushman, 2019-09-08:
> I have the feeling that something is wrong in the SD pins definition referenced above.
> Coud you confirm that your working configuration is based on the following PIN definition ?
> Thierry

I've just confirmed, my pinout is exactly as yours. You can look into my files I provided with my initial post. [https://github.com/Geeetech3D/Diagram/files/3496221/GTM32PRO_VB.zip](url)
 To be clear: I use the SD reader on the LCD controller and not the one on the board itself but they seem to be sharing pins. The SPI2 pins you've mentioned are NOT used by the SD card reader. I think those are used for serial LCDs.
Technically if you are able to compile using my files it should work at least as good (bad) as mine does. Meaning the SD should work. I can't really explain why it would work here but not there other than maybe missing some libraries or something has changed in the code base since I've build it.

- Bushman

thpparent, 2019-09-10:
Agree. Something should have changed since your successful setup.
Could be : 
- the Geeetech LCD2004 hardware (mine is the one with the white back PCB and both connectivity (2x 10 pins + 1x 40 flat)
- Marlin 2.0 library and more specifically the one dedicated to SD

After several attempts, I confirm that refering SPI2 pin out for the external SD reader solved my issue (it works now).
pins_GTM32_PRO_VB becomes : 
#if ENABLED(GTM32_PRO_VB_USE_EXT_SDCARD)
  //
  // SD Card on RepRapDiscount Smart Controller (J2) or on SD_CARD connector
  //
  #define SS_PIN         PB12 / PC11 
  #define SCK_PIN          PB13 // PC12 // PC1
  #define MOSI_PIN         PB15 // PD2  // PD2
  #define MISO_PIN         PB14 // PC8
  #define SD_DETECT_PIN    PC7

#else
  //
  // Use the on-board card socket labeled TF_CARD_SOCKET
  //
  #define SS_PIN           PA4
  #define SCK_PIN          PA5
  #define MOSI_PIN         PA7  
  #define MISO_PIN         PA6 // PA6
  #define SD_DETECT_PIN    -1  // Card detect is not connected
#endif

All files / lib and used in my config are available at : https://drive.google.com/open?id=1X8TSI6Q8rexqwbTk_WlKb63fYAuQSrKh

Did you find a way to address the 3.3V used for the temp ? (new tables ?)

Thank you for your help/support

Thierry

Vertabreak, 2019-11-08:
just to note i did basically get marlin running on the A30 E180 & M201 it needs some work but it does run on the board.

[54 more replies]