Geeetech3D/Diagram #10: Tip: new board  	GT2560_V3.0

kaktusus, 2018-09-21:
The new version of the controller board has a designed LED connector.

I suggest programming DIGILED support, eg **WS2812B**.

This solution will create more possibilities for the user and will be more attractive.
In the code it would be necessary to implement a fixed number of these LEDs _(e.g. 8)_ or to allow the user to specify their number.
Additionally, I will remind you that Gcode instructions allow you to control the colors of lighting. 
:wink:

kaktusus, 2018-09-23:
I would like to draw attention to one more advantage of using this solution.

Once initiated by DIGILED, they shine constantly.
The controller does not have to handle them.
A single control is required for every change in intensity or colour of light.

harlyh, 2018-10-15:
GT2560 rev B inside my A10 also has led connector. However I still haven't found a schematic of RevB board, which would correspond to the actual manufactured board